sinara-hw/STM_Sys_Board #3: LVDS routing

gkasprow, 2023-02-17:
Some LVDS lanes are routed to banks that do not support LVDS at all.
Moreover some of them are routed to banks supplied from 3.3V (bank 0,1,2)
Some banks need resistors to emulate LVDS outputs (0,1,2)
Lanes 8-15 of each slot must be routed to MLVDS transceivers:
[image]
[image]
[image]
[image]
[image]

marmeladapk, 2023-02-28:
Current differential signal compatibility:

| Signal\Slot |	1 |	2 |	3 |	4 |	5 |	6 |	7 |	8 |
| ----- | ----- | ----- | ----- | ----- | ----- | ----- | ----- | ----- |
| 0	 |	:white_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	:white_check_mark: |
| 1	 |	:white_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	❌️ |	:white_check_mark: |	❌️ |
| 2	 |	:white_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |
| 3	 |	:white_check_mark: |	❌️ |	❌️ |	❌️ |	:heavy_check_mark: |	❌️ |	:white_check_mark: |	:white_check_mark: |
| 4	 |	:white_check_mark: |	:white_check_mark: |	❌️ |	:white_check_mark: |	:white_check_mark: |	:white_check_mark: |	:white_check_mark: |	:white_check_mark: |
| 5	 |	:white_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	❌️ |
| 6	 |	:white_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:white_check_mark: |	❌️ |	:white_check_mark: |
| 7	 |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:heavy_check_mark: |	:white_check_mark: |	:heavy_check_mark: |	❌️ |
| 8	 |	 |	 |	❌️ |	:heavy_check_mark: |	 |	 |	 |	 |
| 9	 |	 |	 |	❌️ |	:white_check_mark: |	 |	 |	 |	 |
| 10	 |	 |	 |	❌️ |	:heavy_check_mark: |	 |	 |	 |	 |
| 11	 |	 |	 |	❌️ |	:white_check_mark: |	 |	 |	 |	 |
| 12	 |	 |	 |	:heavy_check_mark: |	:heavy_check_mark: |	 |	 |	 |	 |
| 13	 |	 |	 |	❌️ |	:heavy_check_mark: |	 |	 |	 |	 |
| 14	 |	 |	 |	❌️ |	:white_check_mark: |	 |	 |	 |	 |
| 15	 |	 |	 |	:white_check_mark: |	:white_check_mark: |	 |	 |	 |	 |

:heavy_check_mark:  - full LVDS with I/O

:white_check_mark: - limited LVDS: either output with LVDS25E or input with LVDS io standard

❌️ - not supported

Empty signals are routed to MLVDS transceivers and are supported.

marmeladapk, 2023-10-31:
In terms of I/O we have:

- 40x True LVDS (I/O)
- 40x Emulated LVDS
- 48x SE to LVDS transceivers

True LVDS is the most versatile since it interfaces directly to peripheral modules and can change direction. Then SE to LVDS transceivers are second, while they require writing to a shift register, they also can change signal direction. Emulated LVDS can only be set to input or output at the synthesis time so it's the least versatile.

I analyzed some Sinara modules to assign signals so that if someone were to use them in non-realtime regime it would be the least hassle to do so:

#\Mod | Silpa | HVSUP | DIO | Sampler | Zotino | Urukul | Fastino | Thermostat EEM | Mirny | Zapper
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | O | O | I/O | I | O | O | O | O | O | O
1 | O | O | I/O | I | O | O | O | O | O | O
2 | I | I | I/O | O | I | I | O | I | I | I
3 | O | O | I/O | O | O | O | O | O | O | O
4 | I | O | I/O | O | O | O | O | O | O | I
5 | I | O | I/O | O | O | O | O | O | O | O
6 | O | I | I/O | I | I | O | O | I | O | I
7 | O | I | I/O | O | O | O | I | I | O | O
8 |   |   | I/O | I |   | O |   |   |   |  
9 |   |   | I/O | I |   | O |   |   |   |  
10 |   |   | I/O | I |   | O |   |   |   |  
11 |   |   | I/O | I |   | O |   |   |   |  
12 |   |   | I/O | I |   | O |   |   |   |  
13 |   |   | I/O |   |   | O |   |   |   |  
14 |   |   | I/O |   |   | O |   |   |   |  
15 |   |   | I/O |   |   | O

Most modules have the same I/O on the first 4 signals (SPI).

So my proposal is:

| #    | Type                                                         |
| ---- | ------------------------------------------------------------ |
| 0    | Emulated LVDS, Output SPI CLK                                |
| 1    | Emulated LVDS, Output SPI MOSI                               |
| 2    | True LVDS (I/O), usually Input SPI MISO                      |
| 3    | Emulated LVDS, Output SPI CS                                 |
| 4    | True LVDS (I/O)                                              |
| 5    | True LVDS (I/O)                                              |
| 6    | True LVDS (I/O)                                              |
| 7    | True LVDS (I/O)                                              |
| 8    | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 9    | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 10   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 11   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 12   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 13   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 14   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |
| 15   | SE to LVDS transceiver (slots 1-6) or Emulated LVDS (slots 7, 8) |

This will allow us to use autodetection via I2C and adjust signal direction for most modules without requiring new gateware. Having True LVDS on signal 2. will allow us to use first bank of DIO as an output. Only slots 7 and 8 will be somewhat limited on signals 8-15.

Of course that's assuming that I'll be able to route these connections.

commit 5dc73265 (2024-03-26): PCB routed (there are still DRC errors, but pinout should be final).

- Updated PCB stackup to one from Kasli (6 layers)
- Fixed polarity of 3_PE_CLK
- Updated impedance profiles
- Rerouted most of the PCB
- Fixed #17
- Fixed #3
- Fixed #5
- Fixed #9
- Fixed #12
- Changed IC1 to 24AA025E48T-I/OT, fixes #18
closed 2024-03-26 by marmeladapk